bigjosh/SimpleNeoPixelDemo #5: Not able to display all white

RoelVB, 2017-01-19:
First I want to thank you for the great code.

I'm not able to get all white pixels. For example when I call `showColor(255, 255, 255)` about the first 90 pixels look white, after that they fade to red. (I have a total of 300 pixels)

Same with 255,255,0 (starts with yellow, ends with red) and with 0,255,255  (start with cyan, ends with green).

Is this a bug or just my strip misbehaving?

bigjosh, 2017-01-20:
Sounds like it could be a voltage drop along the strip? Blue and then green go as voltage drops.

Which end are you powering? The one closest to the Arduino? Maybe try connecting an additional 5V power supply to the far end of the string and see what happens?

RoelVB, 2017-01-21:
Whoops. When hooking up a powersupply at the far end I accidentally put 12 volt on it. Now I'm sure the problem is with the strip.

Maybe any advise on a good IP65 WS2812 strip? If not, you can close this case, I feel so stupid.

bigjosh, 2017-01-21:
I've gotten lots of the silicone-sleeved strips that claim to be IP, but none ever are for any real amount of time outside.  If you really need water-tight pixels that last a long time, I'd plan on adding your own weather proofing that is suited to the location and expected service life.

closed 2017-01-21 by bigjosh