bigjosh/SimpleNeoPixelDemo #7: Don't work on my Arduino MEGA 2560

Ldoppea, 2018-01-15:
Hi,

Hi just tried your SimpleNeopixelDemo 'as is' with my Arduino MEGA 2560 and I did not succeed to make it work.

Your code is using PORTB, bit 4, which seems to be the pin 10 on the MEGA. Do you confirm?

When I run the code with rainbow mode, only the first led is lighted and it is strangely blinking. If I remove the wire jump from pin10, all leds finally get a color. 
If I plug the wirejump back, nothing change except the first led blinking. 
If I remove it, all leds get a new color. The color seems to be consistent with the rainbow cycle.

If I use the official adafruit library everything works correctly.

Any idea of what I'm doing wrong?

I'm on windows 10 with Arduino 1.8.5

bigjosh, 2018-01-15:
That sounds like the LEDs are not seeing a delay to give them a change to latch the new color values.

Maybe your LEDs need a longer delay? Can you try changing the value of `RES` here...

https://github.com/bigjosh/SimpleNeoPixelDemo/blob/master/SimpleNeopixelDemo/SimpleNeopixelDemo.ino#L38

...from `6000` to maybe `50000` and see if that helps. Report back with your findings!

Ldoppea, 2018-01-15:
Hi, Thx for your answer

Yes, it was my first thought and I tried 50000 and even 100000 but nothing change. Also tried 500000 but then all leds are tilting white very fast 😆

Ldoppea, 2018-01-15:
Okay, I tried, 50000, 10000, 500000, but I did not try 10000000 which seems to work 🤣 

But now the question is why such a big number?

bigjosh, 2018-01-16:
Not that big of a big number. `10,000,000ns` = `10ms`.  Either (1) you've got some really slow to reset pixels, or (2) something else is going on. Easiest way to see if something else is going on is with an oscilloscope. Got one?

Ldoppea, 2018-01-16:
No I don't, but maybe i'll be able to use one in the next days or weeks. If it is the case i'll try to check this.