bigjosh/SimpleNeoPixelDemo #8: arduino uno - doesn't seem to work with USB unplugged!

ben-xo, 2018-11-25:
I have been adapting your assembler code to generate neopixel signals with interrupts enabled between each pixel. The weird thing is - it works just fine when USB is plugged in, yet as soon as i pull USB out most of the pixels go black (with the occasional one brightening up). It's as if with no USB all the timings are too slow. Any ideas?

Otherwise, with USB in, I can now sample analog AND run neopixels from the same Uno. (it's for a visualiser and i need interrupts between bits/bytes).

bigjosh, 2018-11-26:
Hmmm, when things work when USB is attached it is usually a power/voltage problem. 

What board are you using to make the USB connection?

BTW, no need for interrupts if all you want to do is ADC! The ATMEGA's built-in ADC can do all the sampling work in hardware the background, and have answer ready for you the next time you have time look! Reading the each consecutive result from the ADC can be as short as one instruction!

For more info, check out the ADC section in the datasheet, but the basic idea is to enable to the ADC and then set it to free run , and then read the results out. Setting the ADC to keep results left justified even makes it so you can read a single 8-bit value for each sample which is very fast.

ben-xo, 2018-11-26:
It turned out to be a power problem after all; i had a 7.5v barrel jack plugged in with the neopixels powered from the Vcc pin. When I swapped it round (separate barrel jack on the neopixels, feeding Vcc on the Arduino) everything worked - but now I'm overvolting the neopixels, which seems a bit risky.

The reason I think it's strange, though, is that the AdaFruit NeoPixel library works just fine - that is, the neopixels illuminate just fine - with the 7.5 barrel jack plugged into the Arduino, and the Vcc out to the neopixels.

I can only assume that signalling that fast isn't giving the pins enough time to settle, as the AdaFruit library is slower. It also uses totally different asm instructions, although appears to do basically the same thing.

The reason I needed interrupts was to keep the sampling rate very stable. But you're right, I should be able to read the ADC and have it trigger another sample immediately using the ADC's auto-trigger!

bigjosh, 2018-11-26:
It is hard to imagine how a power change could make one version of the code work and the other not. The AVR clock is the clock, regardless of the voltage - and the timing of the signals generated is based on that clock. If one version of code works and the other does not and the only difference is voltage, then something interesting is going on here!

If you have an oscilloscope, I'd love to see the data output of both SimpleNeoPixel (not working) and Strandtest (working) with all other hardware and connections the same!

ben-xo, 2018-11-26:
Sadly I don't. I'll keep experimenting, though.

closed 2021-07-27 by ben-xo