bigjosh/SimpleNeoPixelDemo #19: Demo sketch (SimpleNeopixelDemo.ino) does not compile for some pins with "impossible constraint in 'asm'" error.

Araris, 2021-04-28:
Hi Josh,

Thank you for the opportunity to work with WS2812 without libraries.
But, when I try to change the control pin, for example,

// Mega D14:
#define WS2812_PORT    PORTJ
#define WS2812_DDR      DDRJ
#define WS2812_BIT         1
or
// Mega A10 (D64):
#define WS2812_PORT    PORTK
#define WS2812_DDR      DDRK
#define WS2812_BIT         2

, in some cases I get a compilation error "impossible constraint in 'asm'".

A similar problem is described in https://forum.arduino.cc/t/passing-port-and-bit-as-parameters-to-asm-block/584236/8 , but, unfortunately, I did not find a solution that I understood there.
I hope for your help, thank you.

bigjosh, 2021-04-28:
This code is specifically for an UNO. It would take significant work to make this code work on boards significantly different than the UNO. See...

[q: this code does not compile for my *not-arduino-uno board*](https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/#:~:text=this%20code%20does%20not%20compile)

closed 2021-04-28 by bigjosh

Araris, 2021-04-28:
Thank you for the quick and comprehensive answer.